ClubRobotInsat/libkicad-robot #3: Crystal design for stm32f103

gbip, 2018-11-14:
We need to find a crystal for the microcontroller and the attached capacitance.
[image]
More information can be found in [this](https://github.com/ClubRobotInsat/kicad/files/2579547/AN2867.Oscillator.design.guide.for.ST.microcontrollers.rev5.2.pdf) application note.

You can find the micro-controller characteristics here :
[image]
*page 105 of [the](https://www.st.com/resource/en/programming_manual/dm00046982.pdf) cortex m4 datasheet*

goowza, 2018-11-15:
The ECS-80-18-5PX (datasheet : [csm-7x.pdf](https://github.com/ClubRobotInsat/kicad/files/2585567/csm-7x.pdf)) is ok.

CL = 18 pF so if Cs = 5 pF, (CL1 // CL2) = 12 pF so  it might works with CL1 = CL2 = 24 pF.

gbip, 2018-11-16:
Thanks a lot @goowza for your answer, I checked your calculations and everything is correct !

 I will add this crystal to the "to buy" list so that we can test it as soon as possible (see #11) ! :smile:

Are you willing to add your comment to the documentation so that people can reuse your work if they need to find an other crystal ? I think that it should go over the [Les différents circuits du Club Robot](https://clubrobotinsat.github.io/doc/elec_mat/circuits/pcb.html) section of the documentation (feel free to add a new entry for crystals). 
@dotsumae is this ok for you ?

gbip, 2018-11-16:
I have troubles finding this crystal on online retailers (such as farnell or rs-components), where can we buy it ?

goowza, 2018-11-17:
Yes i'll do that :)
I can't find it either, but [this one](https://fr.farnell.com/fox-electronics/foxsdlf-080-20/crystal-8-0mhz-20pf-smd/dp/2063972) has almost the same caracteristics (the only differences are the size and the ESR value which is 80 ohms instead of 60 ohms). The gain margin value is changed but is equal to 42,4 so that's still > 5.

gbip, 2018-11-17:
Nice, thanks !