ClubRobotInsat/libkicad-robot #4: Design board protections

gbip, 2018-11-14:
# Designing protections
We should include protection schematics ready to be included in new projects to ensure maximum reliability.

I suggest that every board we design contain this small schematic (the left one) on the power input :
[image] 
[Source](https://electronics.stackexchange.com/questions/373932/protect-a-micro-controller-even-on-mistakenly-connected)

gbip, 2018-11-20:
Is it a good idea to put resistors on each input/output of the micro-controller ? They should limit the current going through them, just like in TP...

gbip, 2018-11-20:
For the coupling capacitors I found ours to be quite big (470 µF is a lot) and it seems that we could use a little bit less.
[Source](https://electronics.stackexchange.com/questions/2262/decoupling-capacitors-what-size-and-how-many)

dotsumae, 2018-11-21:
K for reducing coupling capacitors. 470uF is overkill and even possibly unefficient after some researches. 
I am currently working on the reverse voltage issue using [this video](https://www.youtube.com/watch?v=IrB-FPcv1Dc)