ClubRobotInsat/libkicad-robot #5: Find correct regulator

gbip, 2018-11-14:
Since the stm32f103 needs to be supplied proper 3.3v and that our power supply board only generates 5v we need to find a regulator which will match the following characteristics (to be completed) :
* 5v input
* 3.3v output
* at least capable of outputting 250 mA
* comes in a CMS package

goowza, 2018-11-15:
[This one](https://www.mouser.fr/ProductDetail/Microchip-Technology/TC1262-33VDBTR?qs=sGAEpiMZZMsGz1a6aV8DcEDIzEx6jfI2UkJfaX9b1ao%3d) can do it, and it just need 2 extra capacitors (1 uF) to work.

[Another one](https://www.mouser.fr/ProductDetail/ON-Semiconductor/NCP1117ST33T3G?qs=%2fha2pyFaduj4IwJuyCbpkUdbMhVss5DSvD5YQhE4OjqSspN1A8SdWw%3d%3d) can do the job too, and the output current is up to 1A (seems too good to be true :/). It also need 2 1uF capacitors to work.

gbip, 2018-11-15:
Maybe we can use other characteristics to differentiate the regulators ?
Is the maximum dropout voltage relevant.

closed 2018-11-20 by gbip