ClubRobotInsat/libkicad-robot #8: Logic signal level converter

gbip, 2018-11-14:
The `stm32f103` speaks in 3.3V, but the servomotors speaks in 5V. Even if the current situation kind of works, we should convert the logic level between these two systems.

After a little bit of research I found a transistor `BSS138` that should do the job.
[image]
*[Source](https://learn.sparkfun.com/tutorials/bi-directional-logic-level-converter-hookup-guide/all)*

frenchfrys, 2018-11-15:
Salut on aurait quelques questions à propos du contexte d'intégration du régulateur, notamment ce que prend les servos en courant notamment.

frenchfrys, 2018-11-15:
Voici des possibilités de " Logic Level Shifter", on continue de regarder mais globalement ça à l'air intéressant, à voir si du 25 mA max en sortie pour communiquer avec les servos ça peut passer. 

http://www.ti.com/lit/ds/symlink/sn74lv1t34.pdf :  env 40 cents

https://www.onsemi.com/pub/Collateral/MC74VHC1GT50-D.PDF : env 30 cents

gbip, 2018-11-15:
It is very important to have a component easy to route because it make the PCB simpler and avoid mistakes.
Being easy to solder make the board more reliable because there will be less soldering mistakes.

This is why I would prefer to go over a transistor as it is much simpler to route and to solder because there would be only 3 pins to solder :
* Gnd
* Input (5V)
* Output 

I think that the [BSS-138](https://fr.farnell.com/on-semiconductor/bss138/mosfet-canal-n-50v-220ma-sot-23/dp/2323154?st=bss138) might suit our needs.

smushis, 2018-12-06:
So we thought about doing a bidirectional one with one Mosfet 2n7000 and 2 voltage supply (3.3V and 5V).
To have a 3.3V supply, we will use a linear regulator 5V to 3.3V (LD1117S33TR). We have put the mosfet and the regulator in the shopping list.
[Screenshot Logic Level Converter](https://imgur.com/a/VOx679m)

closed 2019-03-07 by dotsumae