ClubRobotInsat/libkicad-robot #11: To buy list

gbip, 2018-11-16:
This issue is used to track the list of components that will be bought with our next order.

* An [ethernet switch](https://www.ldlc.com/fiche/PB00177504.html)

dotsumae, 2018-11-21:
. [BJT to drive leds](https://fr.farnell.com/diodes-inc/mmbt4401/transistor-npn-40v-0-6a-sot23/dp/2061459)
. [Red SMD led](https://fr.farnell.com/kingbright/kptl-3216surck-01/led-1206-optic-rouge-reverse/dp/2217903)
. [Yellow SMD led](https://fr.farnell.com/kingbright/kptd-3216syck/led-1206-lentille/dp/8530149)
. [Green SMD led](https://fr.farnell.com/wurth-elektronik/156120vs82500/led-vert-1-05cd-573nm-1206/dp/2765005)

gbip, 2018-11-24:
You have to tick it in the top comment...

smushis, 2018-12-06:
- [ ] [Transistor Mosfet 2N7000](https://fr.farnell.com/on-semiconductor/2n7000/mosfet-n-logique-to-92/dp/9845178)
- [ ] [Regulateur 5V to 3.3V LD1117S33TR](https://fr.farnell.com/stmicroelectronics/ld1117s33tr/regulateur-ldo-3-3v-1117-sot/dp/1202826)

gbip, 2019-05-02:
- [ ] [Relay](https://robotdyn.com/relay-module-1-relay-5v.html) x 3

gbip, 2019-05-02:
- [ ] [Lead free solder](https://fr.farnell.com/multicomp/595021/soudure-sans-plomb-rosin-0-7mm/dp/5071306?st=%C3%A9tain%20sans%20plomb)

dotsumae, 2019-05-06:
- [ ] Batteries (peut être plus petites ?) chargeur et témoins 
- [ ] Ecrans raspi
- [ ] Pompes 
- [ ] Vaseline 
- [ ] Trousse secours 
- [ ] Scratch 2 faces 
- [ ] Raccords pneumatiques : thomas
- [ ] R1206 270
- [ ] R1206 560

gbip, 2019-05-11:
- [ ] [Regulators](https://fr.farnell.com/microchip/tc1262-3-3vdbtr/regulateur-ldo-500ma-sot-223/dp/1296594?st=%20TC1262-3.3VDBTR) x10
- [ ] [Mofset NDS351AN](https://fr.farnell.com/on-semiconductor/nds351an/transistor-mosfet-n-sot-23/dp/9845410?st=NDS351AN) x100 (~13€)

ygunday, 2019-05-16:
[ ] - [https://fr.farnell.com/welwyn/w31-r056ja1/resistance-3w-5-0r056/dp/9497170] (Power Resistor) x5

ygunday, 2019-05-16:
- [ ] [Power Resistor if the first option is not available](https://fr.farnell.com/welwyn/w31-r10-ji/resistance-3w-5-0r1/dp/9497200) x5

dotsumae, 2019-05-19:
[ ] [Level-switching MOSFETs](https://fr.farnell.com/microchip/tn0702n3-g/mosfet-20v-0-53a-150-c-1w/dp/2920846) x30

gbip, 2019-05-19:
[ ] - Phoenix connectors (2 and 3)
[ ] - [SOT23 to DIP](https://www.digikey.com/products/en?mpart=PA0083&v=315) (in case of the mosfets do not works, so that we can use BSS138)
Have you bought [this](https://github.com/ClubRobotInsat/libkicad-robot/issues/11#issuecomment-488679438) ?

dotsumae, 2019-05-19:
- [ ] Short Ethernet cables (primewire WD Plus GmbH)