ClubRobotInsat/libkicad-robot #13: Dimension CMS LEDs

gbip, 2018-11-20:
Currently we have a [few](https://clubrobotinsat.github.io/doc/elec_mat/references/ref_composants.html#autres) LEDs in CMS packages left at the club but it is not enough for all the boards.

We need to find LEDs in CMS package (1206 is enough I think) that match the following specification :
* 3.3v supply
* not too much current draw (how much @dotsumae ?)
* 1206 package

We need 3 colors (we still have some red left at the club, see the CMS folder in the "extra" section) : blue, yellow, and green.

dotsumae, 2018-11-21:
I found what we need on farnell : 30 led (red, yellow, green, but not blue (power-hungry color)) for about 7€. I'll order them when I also find the MOSFETs to protect our PCBs

gbip, 2018-11-21:
Can you post the farnell link and add it to #11 ?

dotsumae, 2018-11-21:
Done. The package is not 1206 though, couldn't find it

closed 2018-11-21 by dotsumae