ClubRobotInsat/libkicad-robot #16: Our current connectors (HE14) are unreliable

dotsumae, 2019-03-07:
We should look for new wire to PCB connectors to replace the old HE14. A professor's has suggested to take a look at [this catalog](https://www.phoenixcontact.com/assets/2018/interactive_ed/101_86830/epaper/52007260_EN_HQ_DC_COMBICON_LoR.pdf) from pheonix connectors. Those seem actually easy to use and reliable.

ygunday, 2019-03-14:
This could work: https://www.ebay.fr/itm/Phoenix-Contact-Leiterplatten-Anschlussklemme-MKDSFW-1-5-2-1717091-460/182681587897?hash=item2a88abd0b9:g:4~UAAOSwjJRZdGQI

dotsumae, 2019-05-19:
"Connecteur Phoenix" have been added in [the component list](https://clubrobotinsat.github.io/doc/elec_mat/references/ref_composants.html)

closed 2019-05-19 by dotsumae